kriswiner/BNO055 #1: BNO-055 vs BMX-055 recommendation?

joseangeljimenez, 2015-01-29:
Hello,

First of all, thanks for your awesome work on all the breakout modules you are releasing. I'd like to ask you for a simple side-to-side comparison or recommendation between these two modules. I will be connecting your breakout with an small Arduino variant with only 32KB, no Teensy, and I'd prefer to keep it simple/small on the software side of the ATmega MCU.

Thank you,
Jose Angel

P.S.: I went on to create an issue on your github expecting that this question (and your answer) will surely be of help to others.

kriswiner, 2015-01-29:
If you have only a small amount of memory for programming you might want to use the BNO-055. It is slightly more expensive but has an embedded Cortex M0 processor that uses a pretty sophisticated Kalman-filter-based sensor fusion to output Quaternions and Euler angles directly from register reads. It also has most of the motion-triggered interrupt capability of the BMX-055. You can think of the BNO-055 as  BMX-055 with a dedicated processor for sensor fusion. I think this is the way to go for most people unless you want to write your own sensor fusion for a specific application.

joseangeljimenez, 2015-02-03:
Thanks a lot for your prompt response and advice!