kriswiner/BNO055 #3: Does it need to do mag calibration every time power is up?

howardmah, 2015-09-07:
Hello Kris,

I have been testing BNO055 and found that it's hardware fusion result is very good. The result, however, is good only with good mag calibration. 
When I am testing it, it will be no problem to move the sensor in a 8 figure. However, if it is attached to a big body, like a robot, moving the whole thing in an 8 figure every time it reboots would not be possible. 
Therefore, I tried to save mag offset values and wrote them into the registers directly without calibration. It seemed not working well. It's result was very far from the accuracy with a good calibration.

Is it possible to save mag calibration and reuse it without any new calibration?
If so, could you explain how to do that?

Thanks,

Howard

kriswiner, 2015-09-07:
Hi Howard,

There are two ways. You could simply calculate (measure) the mag biases and
hard code them into your program. Or you could store them in non-volatile
flash memory or EEPROM either on a separate I2C breakout board or on your
microcontroller. Most microcontrollers (like the Teensy 3.x) have some spare
(~2 Kbyte) flash available and you can write small amounts of data there and
retrieve it on start up. Simply writing the values into your program is
surely the easiest method. This is possible since most of the mag
calibration is to compensate for board stresses and other changes to the
magnetometer during assembly; they don't change very much. However, if your
robot has a lot of iron and high-current wires, this might be a problem
since the magnetic environment will be affected by these and you might have
to perform at least one bias calibration with the magnetometer in its
working position; on the robot!

Kris

up? (#3)

<https://github.com/kriswiner/BNO-055/issues/3> .
<https://github.com/notifications/beacon/AGY1qi5rOUCD_m04YZtqEI0Ee7ERiQXoks5
ovUxXgaJpZM4F415m.gif>

chrisspen, 2017-05-29:
I'm running into the same problem. I was y able to store and reload the calibration offsets. Immediately after load, the system flag shows full calibration. However, it seems to be continuously recalibrating, because the system flag, and sometimes magnetometer flag, with sometimes revert back to uncalibrated or partial calibration status.

Is there any way to disable the auto-calibration once you've loaded the correct calibration offsets, so it doesn't effectively lose calibration?

kriswiner, 2017-05-29:
Not that I know of, but check the data sheet (tehy've added more detail on
the calibration) or ask Bosch.

rvincent25, 2018-10-22:
I have been running into the same issue anyone found out how to prevent the IMU to recalibrate?

chrisspen, 2018-10-22:
I save the last known calibration settings, and then re-load it upon power-up.

rvincent25, 2018-10-24:
That's what I do but the calibration flags are still 0 (instead of 3) after reloading the values, so I'm doing something wrong here.