kriswiner/BNO055 #5: No I2C devices found!!

mohit5037, 2016-03-08:
I have teensy 3.2 and BNO055+BMP280. I am trying to connect it with teensy and I have uploaded your code and edited this line

#define AD0 0, so the device address should be 0x28

My connections are as follows:

3.3 V pin of teensy is connected to 3v3 pin of BNO which is adjacent to SCL pin.
Gnd is connected to Gnd Pin of BNO which is adjacent to pin PS0.
SDA and SCL are connected accordingly as shown in the images below.

Scanning...
Unknow error at address 0x01
Unknow error at address 0x02
Unknow error at address 0x03
Unknow error at address 0x04
Unknow error at address 0x05
Unknow error at address 0x06
Unknow error at address 0x07
Unknow error at address 0x08
Unknow error at address 0x09
Unknow error at address 0x0A
Unknow error at address 0x0B
Unknow error at address 0x0C
Unknow error at address 0x0D
Unknow error at address 0x0E
Unknow error at address 0x0F
Unknow error at address 0x10
Unknow error at address 0x11
Unknow error at address 0x12
Unknow error at address 0x13
Unknow error at address 0x14
Unknow error at address 0x15
Unknow error at address 0x16
Unknow error at address 0x17
Unknow error at address 0x18
Unknow error at address 0x19
Unknow error at address 0x1A
Unknow error at address 0x1B
Unknow error at address 0x1C
Unknow error at address 0x1D
Unknow error at address 0x1E
Unknow error at address 0x1F
Unknow error at address 0x20
Unknow error at address 0x21
Unknow error at address 0x22
Unknow error at address 0x23
Unknow error at address 0x24
Unknow error at address 0x25
Unknow error at address 0x26
Unknow error at address 0x27
Unknow error at address 0x28
Unknow error at address 0x29
Unknow error at address 0x2A
Unknow error at address 0x2B
Unknow error at address 0x2C
Unknow error at address 0x2D
Unknow error at address 0x2E
Unknow error at address 0x2F
Unknow error at address 0x30
Unknow error at address 0x31
Unknow error at address 0x32
Unknow error at address 0x33
Unknow error at address 0x34
Unknow error at address 0x35
Unknow error at address 0x36
Unknow error at address 0x37
Unknow error at address 0x38
Unknow error at address 0x39
Unknow error at address 0x3A
Unknow error at address 0x3B
Unknow error at address 0x3C
Unknow error at address 0x3D
Unknow error at address 0x3E
Unknow error at address 0x3F
Unknow error at address 0x40
Unknow error at address 0x41
Unknow error at address 0x42
Unknow error at address 0x43
Unknow error at address 0x44
Unknow error at address 0x45
Unknow error at address 0x46
Unknow error at address 0x47
Unknow error at address 0x48
Unknow error at address 0x49
Unknow error at address 0x4A
Unknow error at address 0x4B
Unknow error at address 0x4C
Unknow error at address 0x4D
Unknow error at address 0x4E
Unknow error at address 0x4F
Unknow error at address 0x50
Unknow error at address 0x51
Unknow error at address 0x52
Unknow error at address 0x53
Unknow error at address 0x54
Unknow error at address 0x55
Unknow error at address 0x56
Unknow error at address 0x57
Unknow error at address 0x58
Unknow error at address 0x59
Unknow error at address 0x5A
Unknow error at address 0x5B
Unknow error at address 0x5C
Unknow error at address 0x5D
Unknow error at address 0x5E
Unknow error at address 0x5F
Unknow error at address 0x60
Unknow error at address 0x61
Unknow error at address 0x62
Unknow error at address 0x63
Unknow error at address 0x64
Unknow error at address 0x65
Unknow error at address 0x66
Unknow error at address 0x67
Unknow error at address 0x68
Unknow error at address 0x69
Unknow error at address 0x6A
Unknow error at address 0x6B
Unknow error at address 0x6C
Unknow error at address 0x6D
Unknow error at address 0x6E
Unknow error at address 0x6F
Unknow error at address 0x70
Unknow error at address 0x71
Unknow error at address 0x72
Unknow error at address 0x73
Unknow error at address 0x74
Unknow error at address 0x75
Unknow error at address 0x76
Unknow error at address 0x77
Unknow error at address 0x78
Unknow error at address 0x79
Unknow error at address 0x7A
Unknow error at address 0x7B
Unknow error at address 0x7C
Unknow error at address 0x7D
Unknow error at address 0x7E
No I2C devices found

BNO055 9-axis motion sensor...
BNO055 Address = 0x28
BNO055 WHO_AM_I = 0x0
BNO055 I AM FF I should be 0xA0
BNO055 ACC I AM FF I should be 0xFB
BNO055 MAG I AM FF I should be 0x32
BNO055 GYRO I AM FF I should be 0x0F
Could not connect to BNO055: 0xFF

kriswiner, 2016-03-08:
If you want to use the edge 3V3 and GND you will have to close the solder jumpers marked 3V3 and GND on the back of the board.

mohit5037, 2016-03-09:
Hi Kris,

Thanks for the early response. I have soldered the jumpers as shown in the image below. I have also soldered jumper for AD0 but I am not giving any voltage to this pin. Now it is finding I2C devices on all addresses. 

[image]

Here are the logs:

I2C device found at address 0x01  !
I2C device found at address 0x02  !
I2C device found at address 0x03  !
I2C device found at address 0x04  !
I2C device found at address 0x05  !
I2C device found at address 0x06  !
I2C device found at address 0x07  !
I2C device found at address 0x08  !
I2C device found at address 0x09  !
I2C device found at address 0x0A  !
I2C device found at address 0x0B  !
I2C device found at address 0x0C  !
I2C device found at address 0x0D  !
I2C device found at address 0x0E  !
I2C device found at address 0x0F  !
I2C device found at address 0x10  !
I2C device found at address 0x11  !
I2C device found at address 0x12  !
I2C device found at address 0x13  !
I2C device found at address 0x14  !
I2C device found at address 0x15  !
I2C device found at address 0x16  !
I2C device found at address 0x17  !
I2C device found at address 0x18  !
I2C device found at address 0x19  !
I2C device found at address 0x1A  !
I2C device found at address 0x1B  !
I2C device found at address 0x1C  !
I2C device found at address 0x1D  !
I2C device found at address 0x1E  !
I2C device found at address 0x1F  !
I2C device found at address 0x20  !
I2C device found at address 0x21  !
I2C device found at address 0x22  !
I2C device found at address 0x23  !
I2C device found at address 0x24  !
I2C device found at address 0x25  !
I2C device found at address 0x26  !
I2C device found at address 0x27  !
I2C device found at address 0x28  !
I2C device found at address 0x29  !
I2C device found at address 0x2A  !
I2C device found at address 0x2B  !
I2C device found at address 0x2C  !
I2C device found at address 0x2D  !
I2C device found at address 0x2E  !
I2C device found at address 0x2F  !
I2C device found at address 0x30  !
I2C device found at address 0x31  !
I2C device found at address 0x32  !
I2C device found at address 0x33  !
I2C device found at address 0x34  !
I2C device found at address 0x35  !
I2C device found at address 0x36  !
I2C device found at address 0x37  !
I2C device found at address 0x38  !
I2C device found at address 0x39  !
I2C device found at address 0x3A  !
I2C device found at address 0x3B  !
I2C device found at address 0x3C  !
I2C device found at address 0x3D  !
I2C device found at address 0x3E  !
I2C device found at address 0x3F  !
I2C device found at address 0x40  !
I2C device found at address 0x41  !
I2C device found at address 0x42  !
I2C device found at address 0x43  !
I2C device found at address 0x44  !
I2C device found at address 0x45  !
I2C device found at address 0x46  !
I2C device found at address 0x47  !
I2C device found at address 0x48  !
I2C device found at address 0x49  !
I2C device found at address 0x4A  !
I2C device found at address 0x4B  !
I2C device found at address 0x4C  !
I2C device found at address 0x4D  !
I2C device found at address 0x4E  !
I2C device found at address 0x4F  !
I2C device found at address 0x50  !
I2C device found at address 0x51  !
I2C device found at address 0x52  !
I2C device found at address 0x53  !
I2C device found at address 0x54  !
I2C device found at address 0x55  !
I2C device found at address 0x56  !
I2C device found at address 0x57  !
I2C device found at address 0x58  !
I2C device found at address 0x59  !
I2C device found at address 0x5A  !
I2C device found at address 0x5B  !
I2C device found at address 0x5C  !
I2C device found at address 0x5D  !
I2C device found at address 0x5E  !
I2C device found at address 0x5F  !
I2C device found at address 0x60  !
I2C device found at address 0x61  !
I2C device found at address 0x62  !
I2C device found at address 0x63  !
I2C device found at address 0x64  !
I2C device found at address 0x65  !
I2C device found at address 0x66  !
I2C device found at address 0x67  !
I2C device found at address 0x68  !
I2C device found at address 0x69  !
I2C device found at address 0x6A  !
I2C device found at address 0x6B  !
I2C device found at address 0x6C  !
I2C device found at address 0x6D  !
I2C device found at address 0x6E  !
I2C device found at address 0x6F  !
I2C device found at address 0x70  !
I2C device found at address 0x71  !
I2C device found at address 0x72  !
I2C device found at address 0x73  !
I2C device found at address 0x74  !
I2C device found at address 0x75  !
I2C device found at address 0x76  !
I2C device found at address 0x77  !
I2C device found at address 0x78  !
I2C device found at address 0x79  !
I2C device found at address 0x7A  !
I2C device found at address 0x7B  !
I2C device found at address 0x7C  !
I2C device found at address 0x7D  !
I2C device found at address 0x7E  !
done

BNO055 9-axis motion sensor...
BNO055 Address = 0x28
BNO055 WHO_AM_I = 0x0
BNO055 I AM FF I should be 0xA0
BNO055 ACC I AM FF I should be 0xFB
BNO055 MAG I AM FF I should be 0x32
BNO055 GYRO I AM FF I should be 0x0F
Could not connect to BNO055: 0xFF

kriswiner, 2016-03-09:
Connect SDA/SCL and 3V3 and GND to the corresponding pins on the Teensy. If
you are using my sketch make sure the SDA and SCL pins in the wire.begin
correspond properly. I think I use pins 16 and 17 be default in the sketch.
Use a voltmeter to measure the voltage between SDA and GND, and between SCL
and GND, and between 3V3 and GND. What are the values?

imag1088 1
<https://cloud.githubusercontent.com/assets/8141787/13626978/bc4fc34e-e5ec-1
1e5-92d1-4a9a78529a85.jpg> 

<https://github.com/kriswiner/BNO-055/issues/5#issuecomment-194132481> .
<https://github.com/notifications/beacon/AGY1qnpKZVWfY1JoAVplfwXhFMl_TEjlks5
prmYcgaJpZM4HrrKK.gif>

mohit5037, 2016-03-09:
Hi Kris,

Here are the voltages:

B/w GND and 3v3 : 3.3V
B/w GND and SCL : 2.89 V
B/w GND and SDA : It fluctuates from 0V to 3V and back-forth.

I have connected SDA to pin 17 of teensy and SCL to pin 16 of teensy.

kriswiner, 2016-03-09:
Good, and in your sketch are you using pins 16 and 217 in the Wire.begin
call?

If you run a scan are you still seeing all of the addresses? This could be
due to a short or something else.

<https://github.com/kriswiner/BNO-055/issues/5#issuecomment-194153291> .
<https://github.com/notifications/beacon/AGY1qmRkymLHGVPCivaxCSLF7AUWScCpks5
prnfogaJpZM4HrrKK.gif>

mohit5037, 2016-03-09:
Yes I am using pin 16 and pin 17 and it is showing I2C device found in all the addresses.

Regarding short I sent you an image above in which I shorted ground, 3v3 and AD0 jumpers.

Here I see one ambiguity, I just unshorted the GND jumper and what I see is this:

GND pin and first rectangle of ground jumper are connected with each other but same is not the case with AD0 pin and first rectangle of AD0 jumper.

So what I want to ask is , should ground pin of sensor and adjacent rectangle of ground jumper should be short internally or not?

[image]

kriswiner, 2016-03-09:
The expected use of this device is to mount directly onto the Teensy with 3V3 and GND taken from the edge pins near the Teensy reset button and connected to the two port on the top of the board. This is the default. You may also take 3V3 and GND from the edge pins for breadboard use. In this case, the solder jumpers on the back marked 3V3 and GND need to be closed to connect the edge pins to the 3V3 and GND circuits, respectively. It looks like you have not done this for GND. Closing the solder jumper means using solder to connect the two rectangular pads together. One of the rectangular pads is connected to either the GND circuit or 3V3 circuit and the other is connected to the edge pin. If the solder also makes a connection between the rectangular pad and the edge pin in these cases it shouldn't matter, but this is not good practice in general.

You have also closed the solder jumper marked ADO. ADO is pulled up by default to set the I2C address at 0x29. If you close the ADO solder jumper this sets the address of the BNO055 to 0x28.

mohit5037, 2016-03-10:
Hi Kris,

Now I have removed solder from every jumper and I am giving it power from those two pins which you mentioned above. Please see the images below for reference but still I am not able to find any of the I2C devices. I have tried running program with both AD0=0 and AD0=1 but still the same result.

I have also checked the shorts with multimeter but none of the jumpers are shorted.

[image]

[image]
[image]
[image]

kriswiner, 2016-03-10:
I don't know what you could be doing wrong. Please send the device back to
me so I can test it and repair it if it is damaged.

Kris

imag1094 1
<https://cloud.githubusercontent.com/assets/8141787/13660555/2a8789f4-e6b1-1
1e5-9860-5bfec850651f.jpg> 

imag1095 1
<https://cloud.githubusercontent.com/assets/8141787/13660562/39f7ff9a-e6b1-1
1e5-8bca-6fd6110228af.jpg> 
imag1096 1
<https://cloud.githubusercontent.com/assets/8141787/13660568/4bae89b6-e6b1-1
1e5-8b51-7f99bf8f3883.jpg> 
imag1097 1
<https://cloud.githubusercontent.com/assets/8141787/13660576/587577fe-e6b1-1
1e5-8c4b-d0fd595d9091.jpg> 

-
<https://github.com/kriswiner/BNO-055/issues/5#issuecomment-194683208> .
<https://github.com/notifications/beacon/AGY1qss3MyUev6Xcw8u8ZN3pEfnYCTx0ks5
pr6_4gaJpZM4HrrKK.gif>