kriswiner/BNO055 #6: Software Yaw not tracking Hardware yaw

duckman01, 2017-04-24:
Hi Kris, Kevin down in Tasmania. I am playing with a BNO055 (Adafruit unit, with wiring for power and SCL,SDA only) and your BNO_055 Nano Basic AHRS code with a Mega  board.
Had some compile issue which l fixed by commenting out lines,981, 993.
I uncommented lines 982, 994 

        uint8_t readByte(uint8_t address, uint8_t subAddress)
{
	uint8_t data; // `data` will store the register data	 
	Wire.beginTransmission(address);         // Initialize the Tx buffer
	Wire.write(subAddress);	                 // Put slave register address in Tx buffer
//	Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection 

alive---------------------------------------------------------------------
	Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive-------------------------------
//	Wire.requestFrom(address, 1);  // Read one byte from slave register address 
	Wire.requestFrom(address, (size_t) 1);   // Read one byte from slave register address 
	data = Wire.read();                      // Fill Rx buffer with result
	return data;                             // Return data read from slave register
}

        void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t * dest)
{  
	Wire.beginTransmission(address);   // Initialize the Tx buffer
	Wire.write(subAddress);            // Put slave register address in Tx buffer
	//Wire.endTransmission(I2C_NOSTOP);  // Send the Tx buffer, but send a restart to keep connection alive--------------------------------------
	Wire.endTransmission(false);       // Send the Tx buffer, but send a restart to keep connection alive---------------------------------------------
	uint8_t i = 0;
//        Wire.requestFrom(address, count);  // Read bytes from slave register address 
        Wire.requestFrom(address, (size_t) count);  // Read bytes from slave register address 
	while (Wire.available()) {
        dest[i++] = Wire.read(); }         // Put read results in the Rx buffer
}
 

and comms issue and changed line 163,164

// Using the BNO055_BMP280 breakout board/Teensy 3.1 Add-On Shield, ADO is set to 1 by default 
//#define ADO 1  //-------------------------------------------
#define ADO 0 //this works for KD--------------------------------------------------------------------------------------

My question relates to your software code outputs, as the hardware Euler angles are as expected and roll tilt correction error to Yaw is +-2 degree and yaw is approx correct heading bearing.
But the software yaw is not working and roll in hardware is pitch in software output.
The software yaw changes when turning IMU but doesn't relate to any direction then returns to what it started at.
You can see in the start results that the Accel/Gyro/Mag say they are Calibrated, but no system Cal.
Also  different revision # and l see you state the software yaw is 180 degrees apposed to the hardware but l am not getting a usable software yaw.

Scanning...
I2C device found at address 0x28  !
done

BNO055 9-axis motion sensor...
BNO055 Address = 0x28
BNO055 WHO_AM_I = 0x0
BNO055 I AM A0 I should be 0xA0
BNO055 ACC I AM FB I should be 0xFB
BNO055 MAG I AM 32 I should be 0x32
BNO055 GYRO I AM F I should be 0x0F
BNO055 is online...
BNO055 SW Revision ID: 3.11
Should be 03.04
BNO055 bootloader Version: 21
accelerometer passed selftest
magnetometer passed selftest
gyroscope passed selftest
MCU passed selftest
Accel/Gyro Calibration: Put device on a level surface and keep motionless! Wait......
Accel/Gyro Calibration done!
Average accelerometer bias (mg) = 
-50.00
-71.00
-16.00
Average gyro bias (dps) = 
-0.06
0.00
0.00
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
Average magnetometer bias (mG) = 
48.13
51.88
-93.75
Not calibrated = 0, fully calibrated = 3
System calibration status 0
Gyro   calibration status 3
Accel  calibration status 3
Mag    calibration status 3
BNO055 initialized for sensor mode....
System Status = 0x5
Sensor fusion algorithm running
ax = 0 ay = 0 az = 0 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
Not calibrated = 0, fully calibrated = 3
System calibration status 0
Gyro   calibration status 3
Accel  calibration status 3
Mag    calibration status 3
Gyro temperature is 18.0 degrees C
                                           Kris code yaw, pitch, roll: 0, 0, 0
                                                          IMU sensor fusion Yaw, Pitch, Roll: 0, 0, 0
System Status = 0x5
Sensor fusion algorithm running
ax = 727 ay = 566 az = -72 mg
gx = 237.19 gy = -135.06 gz = 168.31 deg/s
mx = -43 my = -45 mz = 0 mG
Not calibrated = 0, fully calibrated = 3
System calibration status 0
Gyro   calibration status 3
Accel  calibration status 3
Mag    calibration status 3
Gyro temperature is 18.0 degrees C
                                           Kris code yaw, pitch, roll: -57, -37, 160
                                                          IMU sensor fusion Yaw, Pitch, Roll: 102, 70, 45
System Status = 0x5
Sensor fusion algorithm running
ax = 161 ay = 562 az = 1307 mg
gx = -32.31 gy = 9.31 gz = -111.06 deg/s
mx = -42 my = -43 mz = 0 mG
Not calibrated = 0, fully calibrated = 3
System calibration status 0
Gyro   calibration status 3
Accel  calibration status 3
Mag    calibration status 3
Gyro temperature is 19.0 degrees C
                                           Kris code yaw, pitch, roll: 60, -41, 5
                                                          IMU sensor fusion Yaw, Pitch, Roll: 345, 3, 27

When turning IMU Mag  data doesn't really change much (accel/gyro do) but the hardware yaw/pitch/roll seems to be correct and track a compass rose.
In 1 lot of restarting the system Cal =3, but l never waved the IMU at mag cal time and when l moved IMU after start completion it went to 0.

Below is data showing software and hardware roll/pitch inverted, also accel lost its cal.
I also coded the cal status into the loop, so l could see what was happening 

System Status = 0x5
Sensor fusion algorithm running
ax = 607 ay = -6 az = 822 mg
gx = 0.00 gy = -0.50 gz = -0.12 deg/s
mx = -35 my = -40 mz = -1 mG
Not calibrated = 0, fully calibrated = 3
System calibration status 0
Gyro   calibration status 3
Accel  calibration status 1
Mag    calibration status 3
Gyro temperature is 19.0 degrees C
                                           Kris code yaw, pitch, roll: 125, -37, -0
                                                          IMU sensor fusion Yaw, Pitch, Roll: 36, -1, 37
Any ideas what l have or haven't done correctly ?
Regards Kevin

P/S. I still haven't gone any further with the IMU/GPS my end, any thing happening on the fusion your end?

kriswiner, 2017-04-24:
Hi Kevin,

The magnetometer values are not correct, not sure why but the BNO055 doesn
not put out correct mag values when in the fusion mode as far as I can
figure. I donlt spend a lot of time working with this sensors though, could
be a software error, but I am just reading the mag data registers so not
sure.

Anyway, no need for software heading when you have the heading from the
BNO055 right?

duckman01, 2017-04-25:
Hi Kris thanks for the quick reply. Re the hardware outputs there have been lots of people, myself and my auto pilot designer that have found the BNO055 isn't up to scratch. 
When even a small velocity (walking or in my case 5mph in the boat) is applied to the IMU it looses the plot. There have been numerous threads about this issue and Bosch won't answer any question re this fault.
In my case the IMU yaw will just drift, in Jacks case it also wouldn't update the yaw correctly.
When l first got the BNO055 l thought our prayers had been answered as it had such great tilt compensation <+-2 error but unfortunately in real world testing it isn't up to scratch.
So l was hoping to see how your code worked and not use the fusion.
If the fusion is turned off, maybe your code/magwick could work?
Regards
Kevin

closed 2017-04-25 by duckman01

kriswiner, 2017-04-25:
Hi Kevin,

Well there are a couple of things. You could run the BNO055 in sensor mode
and use the properly scaled sensor data with the Madgwick sensor fusion.
But this isn;t the poblem with the BNO055. The sensors on the BNO055 suck.
It's that simple.

Now you can at least get the most out of them by making sure to calibrate
the accel and gyro offset biases and especially the magentometer response
surface. if in doubt about the latter, see:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

This might help, but you are never going to get great results from the
BNO055 no matter what you do.

In other words, it's the sensor data quality that matters. The best fusion
algorithm in the world cannot make up for crappy data, and that is what you
have in the Bosch accel and mag.

Try what we call the "ultimate Sensor Fusion Solution":

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search

I am sure I recommended this to you before. Maybe it's time you give it a
try?

Kris

Kris

duckman01, 2017-04-25:
Hi Kris, thanks for the info. I thought the BNO sensors were supposed
to have been way better than the others.
Isn't the unit l got of you the same as the ultimate sensor fusion IMU
with the addition of the GPS?.
I had these BNO055 before l got your unit.As well l also had a MPU9250
Chinese copy, this had about +- 7 degree max error versus the Pololu
v3 which had about +-12.Your GPS unit (but using the IMU) on average
had +-7 degree error (same as Chinese MPU9250) although once it did
have +-2 degree for about 2 minutes before it returned to +-7.Also
when l took the GPS unit for a spin in the car at about 20 mph on a
straight bit of road and steering dead straight the GPS heading would
jump up to 5 degrees, which made it unusable as a back up for a yaw
heading (it had full satellite coverage).I never did finish or get to
work (most probable past my level)  code to save cal, so had to
restart cal ever time starting IMU.I currently have good results with
the V3 in a Gimbal, cheating l know, but it is in the range of  Closed
#6 .
 >
 > —
 > , or mute
 > the thread
 > 
 >

thread [2].     

Links:
------
[1]
https://github.com/kriswiner/BNO-055/issues/6#issuecomment-296896570
[2]

duckman01, 2017-04-27:
Kevin Duckmanton17 Carl CourtSwan PointTasmania, 7275043736896903
63944508kduckmanton@iinet.net.au

----- Original Message -----
From: kduckmanton@iinet.net.au
To: "Kris Winer" 
Sent: Thu, 27 Apr 2017 18:03:15 +1000
Subject: Re: [kriswiner/BNO-055] Software Yaw not tracking Hardware
yaw (#6)
 Hi Kris, Bump.Not willing to comment?Also tried the BNO in raw data
mode (AMG) but no usable heading from softwareI take it there has
been no advancement in the GPS fusion?Regards 
Kevin Duckmanton17 Carl CourtSwan PointTasmania, 7275043736896903
63944508kduckmanton@iinet.net.au

----- Original Message -----
From: kduckmanton@iinet.net.au
To:"kriswiner/BNO-055" 
Cc:
Sent:Tue, 25 Apr 2017 15:02:35 +1000
Subject:Re: [kriswiner/BNO-055] Software Yaw not tracking Hardware yaw
(#6)

 Hi Kris, thanks for the info I thought the BNO sensors were supposed
to have been way better than the others.
Isn't the unit l got of you the same as the ultimate sensor fusion IMU
with the addition of the GPS?.
I had these BNO055 before l got your unit.As well l also had a MPU9250
Chinese copy, this had about +- 7 degree max error versus the Pololu
v3 which had about +-12Your GPS unit (but using the IMU) on average
had +-7 degree error (same as Chinese MPU9250) although once it did
have +-2 degree for about 2 minutes before it returned to +-7.Also
when l took the GPS unit for a spin in the car at about 20 mph on a
straight bit of road and steering dead straight the GPS heading would
jump up to 5 degrees, which made it unusable as a back up for a yaw
heading (it had full satellite coverage).I never did finish or get to
work (most probable past my level)  code to save cal, so had to
restart cal ever time starting IMU.I currently have good results with
the V3 in a Gimbal, cheating l know, but it is in the range of  Closed
#6 .
 >
 > —
 > , or mute
 > the thread
 > 
 >

thread [2].       

Links:
------
[1]
https://github.com/kriswiner/BNO-055/issues/6#issuecomment-296896570
[2]

duckman01, 2017-07-21:
Hi  from Tasmania Kris, l have been looking at the first model
Motoin Asset tracker IMU l have and have been getting lower tilt
induced heading error, more like  Closed #6 .
 >
 > —
 > , or mute
 > the thread
 > 
 >

thread [2].     

Links:
------
[1]
https://github.com/kriswiner/BNO-055/issues/6#issuecomment-296896570
[2]

kriswiner, 2017-07-21:
> tkHo0zOZxks5rzXHlgaJpZM4NGzei

Hi Duckman,

There is an API not for the STM32L4 to run the CAM M8Q. I have been doing
some testing of this system described here:

https://hackaday.io/project/25790-asset-tracker

Can't quite remember where we left off...

Kris