kriswiner/BNO055 #7: I2C_MASTER not declared in the scope

shreyamantu, 2017-05-17:
I2C_MASTER and multiple other objects not declared in this scope. Am i missing a header file ?

kriswiner, 2017-05-17:
I have no idea what you are doing.

agademer, 2018-10-22:
I2C_Master is not declared when you comment the teensie lib and try to use Wire.h

Most of "error" lines add a commented version, so I have reverted them and it seems to work.

Original code : 
//  Wire.begin();
//  TWBR = 12;  // 400 kbit/sec I2C speed for Pro Mini
  // Setup for Master mode, pins 16/17, external pullups, 400kHz for Teensy 3.1
  Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

Code for Wire.h
  Wire.begin(); // <= UNCOMMENTED
//  TWBR = 12;  // 400 kbit/sec I2C speed for Pro Mini
  // Setup for Master mode, pins 16/17, external pullups, 400kHz for Teensy 3.1
//  Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400); // <= COMMENTED

etc. 

The question is a bit old, but I thought it might help someone in the future.

Still I'm a bit sorry because you use 2482 bytes of memory and Arduino Uno has only 2048 byte available :p
I will try to see if I can get rid of some intermediary variable... :(

Thanks nonetheless for the great code and explanation associated !