kriswiner/BNO055 #8: Existing Library for the BNO055

jr2629, 2017-09-23:
Hi Kris,

I will be working on a project involving the BNO055 IMU as a device within a larger system, the main MCU for the project is a Teensy 3.5.

Thank you for the code provided here, I was wondering, before I went about making one, is there an existing library format (.cpp + .h file) for this sensor/breakout board (BNO055 + BMP280)?

If it's not something that's currently in place, I will be creating one for just the BNO055 as a stand alone chip.

Thanks,
Joe

kriswiner, 2017-09-23:
No, I don't have one. If I made one I would just be repackaging the
existing functions into the "proper" cpp format. I have done this with a
lot of other sensor libraries but I don't use the BNO055 much so I haven't
gotten around to it yet.

I think adafruit already has an Arduino library, might want to check on
that. I do have one for the BME280 I think. Also Bosch has their API but it
is impenetrable.

See: https://github.com/kriswiner/BME280/tree/master/BME280_Library

Good luck,

Kris
Kris

jr2629, 2017-09-24:
Thanks for the timely reply, just got round to reading through the Adafruit library. 

Did you create this code before they released the library/what was the reason behind your code?

This is gonna be my first Library formatting, so I will run through with it to see how far I can get. Just out of interest which IMU do you commonly use, given that you said you don't use the BNO055?

Joe

kriswiner, 2017-09-24:
Hi Joe,

Bosch's library is the worst kind of spaghetti code, good luck if you start
with that.

I prefer this for accurate absolute orientation:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search

Kris

saedelman, 2017-11-26:
Hello Kris,

I'm wondering if you have any insights into what I am seeing. I have implemented the Madwick update code and I am getting good roll and pitch angles when operating in AMG mode. There is a minimum of jitter, albeit more than what the Bosch fusion code produces in NDOF mode, but my heading indication is not right. It does not appear to be a scaling issue or offset issue. I've adapted your magcalMPU9250 code to determine the bias and scale values for the BNO055 but the heading is not correct nor stable and continues to "walk" after the device is already in a fixed position. When I turn it 90DEG, it may initially only show an increase of 30DEG and then walk back the heading. It does this whether or not I use the bias/scaling values. I'm wondering whether I am feeding the correct w,x,y,z values into the Euler conversion routine to get roll, pitch and yaw, but I am not certain that would manifest itself in this manner. Any insights would be appreciated.

Stephan.

kriswiner, 2017-11-26:
Well, i will assume you have calibrated the sensors, then the likely
problem would be the way you are feeding the sensor data into the Madgwick
filter. Typical convention is to use NED in a consistent manner.

guitchounts, 2021-02-10:
Hi Kris, 

I had a few of your nano-BNO055 boards made and would like to use them with Adafruit's library. Just wiring the boards up the same way as Adafruit's, I get errors relating to the BNO being not found. Is there a workaround that would make it possible to use your board with Adafruit's BNO055 library? Thanks for any help!

kriswiner, 2021-02-10:
Probably just need to get the pin map right.

guitchounts, 2021-02-10:
Sorry, I'm a newb with these things. Does pin map mean how the PCB is wired to the MCU? Or is it the register map definitions? e.g.: 
`#define BNO055_CHIP_ID          0x00`   
`#define BNO055_ACC_ID           0x01`
etc

kriswiner, 2021-02-10:
> etc (or something else?)

Connect MCU SDA to BNO Nano SDA pin, etc. Yes, you have to get the BNO055
I2C address right too...

guitchounts, 2021-02-11:
Here's my setup: I have the nano-BNO055's SCL pin connected to my Teensy 3.2's SCL0; the SDA to SDA0; ground to ground; 3.3V to 3.3V; and the ADR pin grounded via a 10k resistor. I'm finding the I2C address to be 0x28 (using the `i2c_scanner.ino` from [here](https://gist.github.com/tfeldmann/5411375))). 
[image]

The examples in the Adafruit BNO055 library all start (i.e. the sensor is found) but the output doesn't have any sensor readings (e.g. Ada's rawdata_example gives all lines like `X: -0.06 Y: -0.06 Z: -0.06		CALIBRATION: Sys=3 Gyro=3 Accel=3 Mag=3`)

When I run your BNO_055_Nano_Basic_AHRS_t3.ino example (switching I2C address to 0x28), sensor readings appear normal...

I guess I'm wondering if there's something more fundamental how the nano board is wired vs Adafruit's that's making their library not readily compatible with the board? Like, would I need to redefine the register maps or something like that? 

Again, I really appreciate any help!!

kriswiner, 2021-02-11:
I am sure Adafruit is making the assumption that you are using their board.
So maybe theyare expecting a different I2C address or something, who knows.
Please ask Adafruit how their library works...obviously not a hardware
problem.