kriswiner/BNO055 #9: Clarifications on BNO055 and its calibration.

mrrobles09, 2018-03-07:
Good day sir kris winer,

As I was strolling your code here for BNO055 I notice that in your set up for "initBNO055", your OPRMode was modified in CONFIGMODE then set your own values like e.g: Ascale = 2g and Abw =ABW_31_25Hz. After that, you set your OPRMode in NDOF Mode at the end of the routine which NDOF Mode is under fusion mode. 

You've also done the same with getting the calibration for accel/gyro/mag.

**Questions:**
1. How are you sure that the settled values before proceeding to NDOF Mode is not overwritten by its default sensor configuration?
2. How are you also sure that the settled offsets gathered in calibration routines is not overwritten by the sensors auto calibration nature when in NDOF mode?

I've set my calibration before and would like to use this values so that I won't have to recalibrate the sensor every time it is powered on and perhaps use the NDOF feature of BNO055 but this limitations makes my goal hard in getting the navigational position in my project.

Regards,
mrrobles09

kriswiner, 2018-03-07:
1. When using NDOF mode all sensor settings are overwritten, nothing can be
done about that with the BNO055.
2. This is somewhat mysterious, since the NDOF mode also has continuous
calibration and there is no way to turn this off either. So one could
simply choose to use some other mode if you want to choose your own
settings and calibration values and then perform fusion on the host MCU.

mrrobles09, 2018-03-08:
Thanks for the quick reply sir! 

So you mean to say that changing the settings in NDOF or any fusion modes is impossible right? How about the ACC_Config,Gyro_Config, and Mag_Config registers? The data sheet says here that these guys are in "readonly" command and cannot be rewritten? How am I supposed to change the settings when I am in non-fusion modes? Also, is it saving in EEPROM be an option so that I can save my offsets in the sensors internal memory or microcontrollers memory? 

This is my routine in saving my offsets which I followed the steps in the data sheet and then put the routine in my void setup. (page 48, "Setting up Calibration profile")

[image]
[image]

Data sheet link: www.mouser.com/ds/2/783/BST_BNO055_DS000_13-838248.pdf . (page 53, "Register Map page 1")

kriswiner, 2018-03-08:
All I can tell you is what you can read for yourself in the data sheet. In
fusion mode the sensor conditions are fixed. Out of the fusion modes you
can configure the sensors anyway allowed. This is done by writing the
various control registers as per the data sheet.

mrrobles09, 2018-03-08:
Oh okay so about that one. I think that your script here is somewhat questionable especially in its routine where your initBNO055 and Calibration routine for accel/mag/gyro sets up in config mode first then ends with NDOF mode. Also, with regards on how you writeByte the accel,gyro, and mag config,  the data sheet displays that the accel,gyro and mag configs are available only in "readonly" modes and not in "read/write only" modes.

mrrobles09, 2018-03-08:
Sorry for pointing that upfront sir. It seems to me that this observation could lead to better understanding on how should the device work and its specified application to which we user must know for it to be fully use. Thank you sir for providing the knowledge to us beginners and God Bless you!

kriswiner, 2018-03-08:
OK, not sure what you are asking here. The sketches in the repository work
well enough but I am not claiming they are perfect. If you find a better
way of doing it please let me know.

mrrobles09, 2018-03-09:
All I am asking is a clarifications on how does the code work and the following questions below.

1. How are you sure that the settled values before proceeding to NDOF Mode is not overwritten by its default sensor configuration?

2. How are you also sure that the settled offsets gathered in calibration routines is not overwritten by the sensors auto calibration nature when in NDOF mode?

Because as I have read the data sheet for BNO055 it doesn't adds up to what is expected. I'm pointing out the following pictures that made me question what is happening here?

[image]
[image]

**This images is the code for setting up the users own range and is I think a flaw (referring to the highlighted yellow)**

[image]
[image]

**This images are from the data sheet of the BNO055 which is also related to the above images(referring to the highlighted yellow of Acc,Gyro and mag config)**

All I am saying is that with the set up on the code and based on the data sheet, "NDOF fusion mode" is the one being followed which it is auto calibrated and its default sensor config are used. Meaning all those setting up values and offsets/radius are meaningless if you take a good look at it.

kriswiner, 2018-03-09:
Good questions for Bosch.

closed 2018-03-09 by mrrobles09