kriswiner/BNO055 #10: Teensy 3.6?

abdoh747, 2018-09-18:
Hi 
I am trying to use the library with a teensy3.6 and the [BNO055](https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/arduino-code) and it doesnt seem to be producing any output when I open the serial monitor. I am using pins 16 and 17 on the teensy which correspond to SDA0 and SCL0 but still no output. 

I dont get any errors when uploading the BNO_055Nano_Basic_AHRS_t3 script to the teensy board, but 

Any help is appreciated...Thanks

abdoh747, 2018-09-18:
I connected the 3V3 to the pin between 12 and 24, and the GND to the pin just below pin13. The voltage on SDA and SCL is showing 3.28V.

abdoh747, 2018-09-21:
I used this script: https://playground.arduino.cc/Main/I2cScanner
and the board was detecting an i2c device on 0x28....

kriswiner, 2018-09-21:
And are you using 0x28 to read BNO055 data? Can you read the WHO_AM_I
register of the BNO055?

abdoh747, 2018-09-23:
How can I do that?

kriswiner, 2018-09-23:
Huh? What device address are you using to talk with the BNO055?

abdoh747, 2018-09-23:
Sorry, but am a newbie...
I am using a teensy 3.6 running arduino.

abdoh747, 2018-09-23:
Oh sorry, misread your email, 
The address is 0x28 which is what the i2C scanner showed but also what was expected i think...

kriswiner, 2018-09-23:
Is this the address you are using in your sketch?

abdoh747, 2018-09-23:
I am using the BNO_055Nano_Basic_AHRS_t3 sketch, which seems to be using it on line 167

kriswiner, 2018-09-23:
Nope

#define ADO 1
#if ADO
#define BNO055_ADDRESS 0x29 // Device address of BNO055 when ADO = 1