kriswiner/BNO055 #11: BNO055 and MPU9250 "Roll" indication is correct in static position, wrong when moving.

Vliegvogeltla, 2018-10-08:
Hi Kris,
Using the BNO055 in my Attitude indicator instrument project, based on an Arduino Mega. 
The Roll and Pitch info is quite perfect on the test bench.
BUT when installed the instrument in my plane, the Roll indication stays around Zero. 
Used also the MPU9250 chip. Same problem.
Certainly, the problem is related to the acceleration of the plane. 
My question, how can I solve this problem ?
Thanks.

kriswiner, 2018-10-08:
Did you calibrate your sensors? How?

Vliegvogeltla, 2018-10-08:
Hi Kris, 

 

Thanks for your fast reply.

Calibration, done, but it makes no difference, done or not done.

*	Gyroscope:  Standing still
*	Magnetometer: with the figure 8
*	Accelerometer: via the 6 standing positions.

Regards, Marc

 

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

kriswiner, 2018-10-08:
Why do you think the problem is due to the accel of the plane. Maybe you
have a big magnet nearby?

Vliegvogeltla, 2018-10-08:
Hi Kris,

 

In my plane there is for sure no (big) magnet, otherwise the magnetic compass would not operate correctly.

 

The unit is working properly on my test bench, but as soon it’s in my plane, moving forward, then the Roll is not given any indication anymore.

Did the same test with the unit in my car. Standing sill on a ramp (lateral) indication correct, when moving, no indication anymore.

From there my conclusion, a problem with the acceleration. 

 

Same error is reported by Crisau233 (https://electronics.stackexchange.com/questions/324044/imu-roll-angle-correct-when-stopped-wrong-when-moving) with the MPU6050.

Replaced the BNO055 by the MPU6050, same results.

 

Regards, Marc