kriswiner/BNO055 #12: BNO055 hardware fusion is wrong at low angular speed

schadocalex, 2018-10-29:
Hi, I use a BNO055 and I have a strange problem. At low angular speed, the intern kalman does not change the yaw value. If I turn manually the imu at about 3deg/s, the angle and the gyroscopes values returned by the fusion are static (there is some very few noise near 0). But above 5-10 deg/s it's working as expecting (and very well).

If I read raw input (without fusion), gyroscope values become good even at low speed.

**Important note**: I don't use magnetometer because of the huge metallic mass near the imu (the BNO055 is in IMU mode for fusion (acc+gyro)).

kriswiner, 2018-10-29:
I have no idea what is happening here. This sounds like a question for
Bosch...

You might have to do the fusion on the MCU host using the raw sensor
output...

schadocalex, 2018-10-29:
Well, after posting this I did ask Bosch directly (I found a official repo). But we can't delete issues so I let this one. They answered me, you can see the answer if you want: https://github.com/BoschSensortec/BNO055_driver/issues/12#issuecomment-433944642

Sorry for the trouble.

closed 2018-10-29 by schadocalex

kriswiner, 2018-10-29:
It is going to be hard (impossible really, without significant drift) to
get accurate heading without a magnetometer but if all you care about is
roll and pitch then Madgwick's 6 DoF fusion algorithm works well with
well-calibrated accel/gyro data with even a modest MCU.

https://github.com/kriswiner/MPU6050/blob/master/MPU6050Library/MPU6050IMU.ino

schadocalex, 2018-10-29:
I have a very special case: I care only about relative yaw (relative since an old yaw position). I only want to know my position relative to a previous one, so I think with gyro it will be ok (I will comment with my tests results If you want).

The heading in my case will be given by others sensors than a magnetometer (that does not work in my case, I have 2t of steel and high DC power (about 10kW sometimes) near the imu).