kriswiner/BNO055 #14: Any problem with software yaw angle

brightproject, 2024-06-17:
Hello mr. @kriswiner 
I use `BNO055` and this sketch

> https://github.com/kriswiner/BNO055/blob/master/BNO_055_Nano_Basic_AHRS_t3.ino

The resulting roll and pitch angles from the `BNO055` fusion and those calculated by the `Madgwick` or `Mahony` algorithm are similar.
But the yaw in `Hardware` is displayed according to turns, and in `Software` it is yaw.
Commenting a line:

> // yaw *= 180.0f / PI;

Doesn't give results... because... there was a suspicion that yaw is somehow calculated in `radians/sec`.
```
void readMagData(int16_t * destination) {
 uint8_t rawData[6]; // x/y/z gyro register data stored here
 readBytes(BNO055_ADDRESS, BNO055_MAG_DATA_X_LSB, 6, &rawData[0]); // Read the six raw data registers sequentially into data array
 destination[0] = ((int16_t)rawData[1] << 8) | rawData[0] ; // Turn the MSB and LSB into a signed 16-bit value
 destination[1] = ((int16_t)rawData[3] << 8) | rawData[2] ;
 destination[2] = ((int16_t)rawData[5] << 8) | rawData[4] ;
} // readMagData
```

Output data
```
System Status = 0x5
Sensor fusion algorithm running
ax = 7 ay = -39 az = 1011 mg
gx = 0.00 gy = 0.06 gz = 1.37 deg/s
mx = 0 my = 66 mz = 8 mG
qx = 0.71 qy = 0.01 qz = 0.01 qw = -0.71
Gyro temperature is 41.0 degrees C
Software Roll, Pitch, Yaw: 0.14, -1.09, -1.57
Hardware Roll, Pitch, Yaw: 1.31, -2.06, 57.19
Hardware x, y, z linear acceleration: 15.00, -57.00, 9.00
Hardware x, y, z gravity vector: -35.00, 22.00, 979.00
rate = 307.44 Hz
```
The magnetic strength of the X axis is always 0, no matter how I rotate the board with the sensor.
The magnetometer sensor itself is fine, I ran the adafruit code and everything appears fine.
[image]
I would appreciate your clarification if you find the time.

kriswiner, 2024-06-17:
If the x-axis mag data is always 0 (do the other axis mag values change?)
but the mag is working fine (verified by other means)  then the sketch is
not reading the magnetometer properly. I do not know why, could be a lot of
reasons.

Wrong I2C address, improper calibration, some error in the sketch (although
it has worked fine for me and others)....

That particular sketch is intended to run on a Teensy 3.2...is that what
you are using? If not, then the I2C API is likely the problem.

brightproject, 2024-06-17:
> Wrong I2C address, improper calibration, some error in the sketch (although it has worked fine for me and others)....

Yes, I agree.
Any reason is possible, from modification of the sensor itself, to the specific device on which the code is running.
I use `stm32f411.`
And by the way, while I was writing this essay, I was doing the “figure eight” movement with a board with a sensor and... the mx axis suddenly showed the `value 78 mG`.
And in general I noticed that the period for updating the magnetometer values ​​is about 10-15 seconds.
Although all other values ​​are updated according to those set here
```
#define PERIOD_UPDATE (100)
if (delt_t > PERIOD_UPDATE) { // update LCD once per half-second independent of read rat
 // check BNO-055 error status at 10 Hz rate
}
```
Adafruit used this feature
> https://github.com/rickkas7/Adafruit_BNO055_RK/blob/48e855b6da1e6aec86c83dce8b582a8a0b0d09f8/src/Adafruit_BNO055.cpp#L633

```
    /* Magnetometer offset range = +/- 6400 LSB where 1uT = 16 LSB */
    offsets_type.mag_offset_x =
        (read8(MAG_OFFSET_X_MSB_ADDR) << 8) | (read8(MAG_OFFSET_X_LSB_ADDR));
    offsets_type.mag_offset_y =
        (read8(MAG_OFFSET_Y_MSB_ADDR) << 8) | (read8(MAG_OFFSET_Y_LSB_ADDR));
    offsets_type.mag_offset_z =
        (read8(MAG_OFFSET_Z_MSB_ADDR) << 8) | (read8(MAG_OFFSET_Z_LSB_ADDR));
```

kriswiner, 2024-06-17:
This I2C API:

#include <i2c_t3.h>

is written specifically for the Teensy 3.2 and will not work for the STM32,
or will not work well....

So if you are seeing strange results, that is likely the reason.

brightproject, 2024-06-17:
> This I2C API: #include <i2c_t3.h> is written specifically for the Teensy 3.2 and will not work for the STM3

I'm used only Wire.h
```
// #include <i2c_t3.h>
// #include <SPI.h>
#include <Wire.h>
```

kriswiner, 2024-06-17:
The question is, what Wire.h. ANyway, I would suspect problems here. Either
the API has buffer limitations, extra delays, race conditions, or some
other problem.

Why not just use the Adafruit BNO055 library and be done with it?

closed 2024-06-17 by brightproject